cotestatnt/AsyncTelegram2 #90: Messages not sent

pablopeu, 2022-11-26:
Hi Tolentino.
Had to reinstall my whole development computer so I went straight to the new 2.02 Arduino IDE, downloaded latest board definitions for esp8266/esp32, AsyncTelegram2 and wifimanager (downloaded directly from library and board managers)
 
This example is almost the same as one we debugged time ago for another issue, but now if I send a message from Telegram, its received by the board but the reply never reaches Telegram.

I'm lost, because this example is a subset of a full program that used to work, and now every old evolution of it does the same.

here is the code:

`
#include <AsyncTelegram2.h>
#include <time.h>

// Timezone definition
#define MYTZ "UTC+3"  //timezone https://sites.google.com/a/usapiens.com/opnode/time-zones

#include <WiFiManager.h>  //https://github.com/tzapu/WiFiManager

#ifdef ESP8266
BearSSL::WiFiClientSecure client;
BearSSL::Session session;
BearSSL::X509List certificate(telegram_cert);

#elif defined(ESP32)
#include <WiFiClientSecure.h>
WiFiClientSecure client;
#endif

AsyncTelegram2 myBot(client);
// Telegram BOT Token (Get from Botfather)
const char* token = "5680874063:AAFxEE8djQ0pn1ezs2rbvPp7RCWK7yDr1-k";  // Telegram token
#define nombrecamara "Test base Telegram"                              //para OTA y Wi-Fi
int64_t userid = 4404433;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Arrancando...");
  delay(50);
  String message = "";  //variable local

  // inicio de configuracion wifi
  WiFi.mode(WIFI_STA);
  WiFi.hostname(nombrecamara);
  WiFiManager wm;  //WiFiManager local initialization
  wm.setTimeout(60);
  wm.autoConnect(nombrecamara);
  // fin configuracion wifi

#ifdef ESP8266
  // Sync time with NTP, to check properly Telegram certificate
  configTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
  time_t now = time(nullptr);
  //Set certficate, session and some other base client properies
  client.setSession(&session);
  client.setTrustAnchors(&certificate);
  client.setBufferSizes(1024, 1024);
#elif defined(ESP32)
  // Sync time with NTP
  configTzTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
#if USE_CLIENTSSL == false
  client.setCACert(telegram_cert);
#endif
#endif

  // Set the Telegram bot properies
  myBot.setUpdateTime(2000);
  myBot.setTelegramToken(token);

  // Check if all things are ok
  Serial.print("\nTest Telegram connection... ");
  myBot.begin() ? Serial.println("OK") : Serial.println("NOK");

  Serial.print("Bot name: @");
  Serial.println(myBot.getBotName());

  message = "ARRANQUE MONITOR: ";
  message += nombrecamara;
  message += "\n";
  message = "AYUDA: Ingresa /ayuda para obtener la lista de comandos";
  message += "\n";
  despachar(message);
  Serial.print("ARRANQUE MONITOR: ");
  Serial.println(nombrecamara);
}

void loop() {
  String msgText = "";
  TBMessage msg;  // hay mensajes de Telegram?
  if (myBot.getNewMessage(msg)) {
    msgText = msg.text;
    Serial.print("texto: ");
    Serial.println(msgText);
    if (msgText.startsWith("/config")) msgText = confi();
    if (despachar(msgText)) Serial.println("Mensaje enviado correctamente");
    msgText = "";
  }

}  // FIN DEL PROGRAMA

// return true if message is successful delivered
bool despachar(String& theMessage) {
  TBMessage msg;
  msg.sender.id = userid;
  return myBot.sendMessage(msg, theMessage);
}

String confi() {
  String message = "CONFIGURACIÓN: ";
  message += nombrecamara;  //normally is a variable
  message += "\n";
  message += "\n";
  message += "La frecuencia de las notificaciones es de: ";
  message += 100;  //normally is a variable
  message += " horas";
  message += "\n";
  message += "La alarma de baja temperatura esta configurada en: ";
  message += 2;  //normally is a variable
  message += "°C";
  message += "\n";
  message += "La alarma de alta temperatura esta configurada en: ";
  message += 30;  //normally is a variable
  message += "°C";
  message += "\n";
  message += "La frecuencia de las alarmas es de: ";
  message += 10;  //normally is a variable
  message += " minutos";
  message += "\n";
  return message;
}

`

On the serial monitor I only get the acknowledgement of the message received, but nothing goes back to Telegram.

[image]
[image]

pablopeu, 2022-11-26:
Debug output:

`

[D][AsyncTelegram2.cpp:296] getNewMessage():	
{
  "update_id": 639914481,
  "message": {
    "message_id": 261,
    "from": {
      "id": edited,
      "is_bot": false,
      "first_name": "edited",
      "last_name": "edited",
      "username": "edited",
      "language_code": "en"
    },
    "chat": {
      "id": edited,
      "first_name": "edited",
      "last_name": "edited",
      "username": "edited",
      "type": "private"
    },
    "date": 1669505556,
    "text": "/config",
    "entities": [
      {
        "offset": 0,
        "length": 7,
        "type": "bot_command"
      }

texto: /config

[D][AsyncTelegram2.cpp:525] sendMessage():	
{
  "chat_id": 461168382047edited,
  "text": "CONFIGURACIÓN: Test base Telegram\n\nLa frecuencia de las notificaciones es de: 100 horas\nLa alarma de baja temperatura esta configurada en: 2°C\nLa alarma de alta temperatura esta configurada en: 30°C\nLa frecuencia de las alarmas es de: 10 minutos\n",
  "parse_mode": "HTML"
}

[E][AsyncTelegram2.cpp:207] getUpdates():	{"ok":false,"error_code":400,"description":"Bad Request: chat not found"}
[D][AsyncTelegram2.cpp:190] getUpdates():	Connection closed from server
[D][AsyncTelegram2.cpp:38] checkConnection():	Start handshaking...
[D][AsyncTelegram2.cpp:47] checkConnection():	Connected using Telegram hostname
Last connection was 290 seconds ago
`

pablopeu, 2022-11-27:
Tried also adding the bot to a group as admin and I get the same debug:

[E][AsyncTelegram2.cpp:207] getUpdates(): {"ok":false,"error_code":400,"description":"Bad Request: chat not found"}

cotestatnt, 2023-03-16:
Hi @pablopeu, I'm sorry for the big late reply, but I've been very busy in recent months.

Do you still have this problem or did you manage to solve it?
The problem is that you are using `msg.sender.id` instead of `msg.chatId` probably due to some old example forgotten somewhere.

The `msg.sender.id` property is indicative of the user who sent the message (to which you may want reply), but the recipient of the outgoing message **must be indicated with the msg.chatId property**. 
This is necessary because in this way the library can handle "normal" users and groups in the same way and the bot can send messages to both (the only thing that changes between a group and a user is the negative id in case of groups)

However, if you need to send a message to a specific user or group it is preferable to use the method 
```
sendTo(chatid/userid, message)
``` 
so you don't have to make intermediate copies of the variables.

pablopeu, 2023-03-16:
Hi Tolentino, thanks for the reply. It puzzled me for a long time, but this last weekend I was commited to solve it, so again I went to study your examples and figured out the **sendTo(chatid/userid, message)** was theway to go, made the slight modifications in the code and all went as expected.

I do have a new puzzle regarding asynctelegram2/thingspeak library but thats for a new post once I can write a good example case.

Thanks again for your help and the work you do with the library and its support

closed 2023-03-16 by pablopeu