cotestatnt/AsyncTelegram2 #94: Frequent reset after update to 2.1.7

MudiStar, 2023-01-01:
Hello 

I'm using your library in several esp 8266 and I'm pretty happy with it.
So my code was running fine but since I updated to 2.1.7, I get the following reset sometimes 3 times per hour:
Sometimes it takes longer.

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0007b6e0
~ld

I also recognized, that if I don't have internet through the connected wifi (router restart after power outage) the esp keeps resetting with the same error.

Do you need more infomation to debug the problem? 

Thanks

MudiStar, 2023-01-01:
I was just able to record another reset:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3ffff4b0 end: 3fffffb0 offset: 01a0
3ffff650:  3fff0694 3fff6a24 401003ad 00000000  
3ffff660:  0016fa85 3fff4168 00004181 000000fd  
3ffff670:  59ef9205 4bc6a7f0 00000000 00003a98  
3ffff680:  59ef9205 00000000 4bc6a7f0 00000000  
3ffff690:  000000fd 000000fd 4010037e 00000000  
3ffff6a0:  59ef9205 3ffff6d0 3fff425c 40213b4a  
3ffff6b0:  00000000 3fff5504 3fff0fe4 00003a98  
3ffff6c0:  00000000 0005a69a 3fff0fe4 40208a00  
3ffff6d0:  000000fd 59f8b3b1 58d5a2a5 3c4bdcd3  
3ffff6e0:  00000008 00000001 0016fa84 00000000  
3ffff6f0:  40239504 40239554 3fff425c 40238b42  
3ffff700:  3fff475c 3ffe98ca 3fff425c 3ffe98ca  
3ffff710:  00000000 00000001 3fff0fe4 40208b1f  
3ffff720:  00000000 00000001 3fff0fe4 40208d0f  
3ffff730:  00000019 3fff03d0 00000001 40213a78  
3ffff740:  00000000 00000000 000001bb 4021418a  
3ffff750:  00000000 3fff5c2c 3fff0fe4 40207bd2  
3ffff760:  3fff4164 4bc6a7f0 589374bc 3ffe99c0  
3ffff770:  000001bb 3ffe98ca 3fff0fe4 3ffe99c0  
3ffff780:  000001bb 3ffe98ca 3fff0fe4 40208db1  
3ffff790:  4021d268 dca79a95 4021d268 dca79a95  
3ffff7a0:  3ffff810 00000000 3ffefe4c 4021b9ae  
3ffff7b0:  3ffff810 00000000 3ffefe4c 4020b4e8  
3ffff7c0:  00000000 00000000 3ffefe4c 4020b553  
3ffff7d0:  0000000c 00000000 401003ad 00000000  
3ffff7e0:  00000000 3fff0fe4 3ffefe4c 00000000  
3ffff7f0:  00000000 00000010 3ffefe4c 3fff0368  
3ffff800:  00000000 00000000 3ffefe4c 4020b6c4  
3ffff810:  696c227b 2274696d 222c313a 656d6974  
3ffff820:  2274756f 222c303a 7366666f 3a227465  
3ffff830:  50007d30 401043b8 00000005 4010222c  
3ffff840:  00000005 00000000 00000020 40100298  
3ffff850:  40102db7 3ffee130 00000005 4010222c  
3ffff860:  3ffea745 401054c3 3ffee130 3ffee8c0  
3ffff870:  00000005 00000000 00000020 40100298  
3ffff880:  00000012 59931c86 00000005 4010222c  
3ffff890:  3ffea745 401054c3 3ffee108 40100298  
3ffff8a0:  40102db7 3ffee108 3ffee8c0 4010222c  
3ffff8b0:  00000000 00000000 0000001f 40100298  
3ffff8c0:  3ffeafd8 00000000 3fffc228 40105d5d  
3ffff8d0:  4000050c 59b4d9f1 4010344e 00000100  
3ffff8e0:  4000dd1a 00000030 00000010 ffffffff  
3ffff8f0:  4000066d 59bcc1c7 4bc6a7f0 000059bc  
3ffff900:  00004bc6 7438f12a 00000010 fffffffe  
3ffff910:  00007438 3fffc6fc 7f1e7b90 59bcc1c7  
3ffff920:  4bc6a7f0 3ffefe4c 3fff0368 00000030  
3ffff930:  401003b8 00000030 3fffc228 40105d5d  
3ffff940:  4000050c 0016f700 00000000 00000000  
3ffff950:  4020ebf3 00000030 00000010 ffffffff  
3ffff960:  4020ebf0 0016f72c 00418937 3ffefe4c  
3ffff970:  00004bc6 00000000 00000010 fffffffe  
3ffff980:  00000000 3fffc6fc 00000000 00000000  
3ffff990:  3ffffd60 3ffefe4c 3fff0368 00000030  
3ffff9a0:  3ffeafe4 2c9f0300 4000050c 3ffefe54  
3ffff9b0:  00000000 3fff33dc 3ffffb78 4020da9a  
3ffff9c0:  3ffffb78 3fff424f 00000006 000007ed  
3ffff9d0:  000006d9 000006d9 3ffe86d4 40100973  
3ffff9e0:  3ffffb78 32373631 34333036 00000006  
3ffff9f0:  00000000 4bc6a7f0 08b43958 1a91350a  
3ffffa00:  00000000 00000000 4bc6a7f0 00000000  
[... 123 lines cut ...]
0x40100b52: free(void*) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 595
0x4020ec1b: AsyncTelegram2::getNewMessage(TBMessage&) at C:\Users\----\Documents\Arduino\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 206
0x40100298: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100298: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100298: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100298: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x402126b5: String::invalidate() at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 295
0x402016aa: TBMessage::~TBMessage() at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 79
0x40100b7c: malloc(size_t) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 821
0x401004d8: pvPortMalloc(size_t, char const*, int) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc/umm_heap_select.h line 85
0x4020166a: TBUser::~TBUser() at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/WString.h line 79
0x40100298: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100298: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
0x40100298: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 181
[... cut ...]
0x40216c16: esp8266::polledTimeout::timeoutTemplate   >::expiredOneShot() const at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/PolledTimeout.h line 234
0x401003ad: millis() at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 193
0x40213b98: loop_wrapper() at C:\Users\----\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 201

MudiStar, 2023-01-09:
So it got a lot better since the esp core update to 3.1.0
Still one problem remains: reset of the esp if no internet is available.

0x4020ac08: AsyncTelegram2::checkConnection() at C:\Users\Arduino\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp line 39

What I understand is, that the connection is checked in the frequency of the main loop and then the esp crashes 
this is what I also see in the serial monitor

1) all startet with begin():

bool AsyncTelegram2::begin()
{
    checkConnection();
    return getMe();
}

2) Connection is checked:
bool AsyncTelegram2::checkConnection()
{
    // Start connection with Telegramn server (if necessary)
    if (!telegramClient->connected())
    {
        m_lastmsg_timestamp = millis();
        log_info("Start handshaking...");

        if (!telegramClient->connect(TELEGRAM_HOST, TELEGRAM_PORT))
        {
            Serial.println("\n\nUnable to connect to Telegram server");
            reset();
        }

3) reset is called

bool AsyncTelegram2::reset(void)
{
    log_info("Restart Telegram connection\n");
    telegramClient->stop();
    m_lastmsg_timestamp = millis();
    m_waitingReply = false;
    return checkConnection();
}

4) calls checkConnection: endless loop

MudiStar, 2023-01-13:
I'm running into resets on the ESP8266 again, like 15 times in 24h. The ESP placed next to the router
Everything works fine on the ESP32.

@cotestatnt 
Is there a way you can help me out? 
I recorded more Exception Stacks for debugging.

cotestatnt, 2023-01-16:
Hi @MudiStar 
First of all, I'm sorry for late reply but in this last months I was really busy at work.

I suspect this bug is actually "double".
The reset due to the reset() statement (sorry for the pun) should be caused by a "death loop" that this statement introduces.

As for the reset which occurs once every 15h, I think it is due to the "delay" introduced by the failed connection with the Telegram server which can happen from time to time for the various reasons that trigger the ESP8266 software watchdog.

I'm trying out some recently introduced changes. As soon as I have positive results I also update the repository.

cotestatnt, 2023-01-16:
Hi @MudiStar 
I just uploaded the [edited file](https://github.com/cotestatnt/AsyncTelegram2/commit/41c3b585abb1a3094362abe091931470c153be06).
I've been trying a sketch since this morning and it hasn't reset until now, looks promising.

To make it easier for me to manage this problem, I make sure that on reboot the esp sends me a message containing the cause of the last reset

```C++
extern "C" {
#include <user_interface.h> // https://github.com/esp8266/Arduino actually tools/sdk/include
}

void setup {
  .....
  ......
  rst_info *myResetInfo;
  myResetInfo = ESP.getResetInfoPtr();

  char welcome_msg[256];
  snprintf(welcome_msg, sizeof(welcome_msg),
    "myResetInfo->reason %x \nBOT @%s online\n/help all commands avalaible.",
    myResetInfo->reason,
    myBot.getBotName()
  );

  // Send a message to specific user who has started your bot
  myBot.sendTo(userid, welcome_msg);
}
```

MudiStar, 2023-01-16:
Hello @cotestatnt 

I'm happy that you "are back" ;-) 

I compiled my two programs I'm using your library with. Fingers crossed.
In the meantime, I've noticed that it seems to be stable just running your OTA example.
If I add a webserver and some other non-blocking functions, everything starts getting unstable. Same code on the ESP32 works fine.

My WakeOnLan code is pretty straight forward. So no pin readings or something
See attached picture. I couldn't find a pattern yet.

[image]

MudiStar, 2023-01-16:
is is not working for me. two restarts so far
I try to get a stack dump

MudiStar, 2023-01-16:
This doesn't happen if I'm just running the Webserver
Maybe you know what is going on

Exception 5: Alloca: MOVSP instruction, if caller's registers are not in the register file
PC: 0x40100217
EXCVADDR: 0x00000000

Decoding stack results
0x402166b0: run_scheduled_recurrent_functions() at C:\Users\\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\Schedule.cpp line 221
0x40204c1d: Login() at C:\Users\\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 597
0x40213b0c: esp8266::MDNSImplementation::MDNSResponder::_udpAppend16(unsigned short) at C:\Users\\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1186

MudiStar, 2023-01-17:
Reset is triggered by Telegram connection:

12:49:55.996 -> 12:49:56   17:1:2023   Winterzeit
12:49:56.999 -> 12:49:57   17:1:2023   Winterzeit
12:50:12.091 -> 
12:50:12.091 -> 
12:50:12.091 -> Unable to connect to Telegram server
12:50:20.768 -> 
12:50:20.768 ->  ets Jan  8 2013,rst cause:4, boot mode:(3,7)
12:50:20.768 -> 
12:50:20.768 -> wdt reset

MudiStar, 2023-01-17:
Each time a little different.
I stop spamming now 

Exception 5: Alloca: MOVSP instruction, if caller's registers are not in the register file
PC: 0x401003df
EXCVADDR: 0x00000000

Decoding stack results
0x4021135c: Stream::SendGenericPeekBuffer(Print*, int, int, unsigned int) at C:\Users\\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\StreamSend.cpp line 66
0x402195dc: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSAnswer_PTR_NAME(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1665
0x401003df: millis() at C:\Users\\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266\core_esp8266_wiring.cpp line 176

cotestatnt, 2023-01-19:
Good morning @MudiStar 
I've just noticed you are running the latest esp8266 core version while mine was a little older.
I'm triyng now with the latest too.

Unfortunately with the esp8266 there has always been this problem and I have never been able to really figure it out because I mainly use the esp32.
I'll update you as soon as I have some more information about the issue.

MudiStar, 2023-01-19:
I have an ESP8266 running without any problems since October with your version 2.1.6
after fixing a bug in just one line with an additional "if", it keeps crashing randomly
The same line is ok running on the ESP32. 

Maybe a library was updated simultaneously or I was just lucky that it was working
Made no sense  for me

If you need my code or something, feel free to ask