cotestatnt/AsyncTelegram2 #100: How to enable serial log debug

sanderkob, 2023-03-22:
I would like to get debug output.
System: ESP8266 version 3.1.1
Test program: example echobot

In echobot.ino, I set 
#define DEBUG_ENABLE 1
#define USE_CLIENTSSL false 

The program runs fine, but I do not get debug info in the Serial output, merely:
	Test Telegram connection... OK
	Message from @xxxxxxx:
	test

What can I do to get debugging info in the serial output?

cotestatnt, 2023-03-23:
Hi @sanderkob 
You need to set this #define to 1

https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.h#L25

sanderkob, 2023-03-23:
Thank you, it works.
I was not aware that in Arduino IDE any #define in the sketch is not  seen by the header file.
On a related issue,
what is the use of the statement
#ifndef DEBUG_ENABLE
in line 24 of asynctelegram2.h? I think lines 24 and 26 can be deleted.

cotestatnt, 2023-03-28:
> what is the use of the statement

I wrote it this way because I was wondering if I could find a way to define DEBUG_ENABLE inside the sketch, instead in library sources but then I had other things to do and I didn't spend any more time on it.

closed 2023-03-28 by sanderkob