cotestatnt/AsyncTelegram2 #101: Firmware update fails to restart

pablopeu, 2023-03-26:
Hi @cotestatnt , using your example for the firmware update I managed to make it work in my program, well almost, firmware updates correctly, but after finishing it fails to restart the ESP8266:

[image]

I can see the progress in serial monitor:

`
Cargado 532480 de 548736 bytes...
Cargado 540672 de 548736 bytes...
Cargado 548736 de 548736 bytes...
Flash Completo!

Unable to connect to Telegram server
`

And after a few seconds its unable to connect to the server thus not reaching the point in the program where it restarts I guess. This is the code where it shows the flash complete, its your code but translated to spanish:

`
  t_httpUpdate_return ret = UPDATER.update(client, file_path);
  Serial.println("Flash Completo!"); // this is the last activity
  client.stop();

  switch (ret) {
    case HTTP_UPDATE_FAILED:
      report = "HTTP_UPDATE_FALLIDO. Error (";
      report += UPDATER.getLastError();
      report += "): ";
      report += UPDATER.getLastErrorString();
      myBot.sendTo(userid, report.c_str());
      break;

    case HTTP_UPDATE_NO_UPDATES:
      myBot.sendTo(userid, "HTTP_UPDATE_NO_UPDATES");
      break;

    case HTTP_UPDATE_OK:
      myBot.begin();
      myBot.sendTo(userid, "UPDATE OK.\nReiniciando en unos segundos...");
      // Wait until bot synced with telegram to prevent cyclic reboot
      while (!myBot.noNewMessage()) {
        Serial.print(".");
        delay(50);
      }
      ESP.restart();

      break;
    default:
      break;`

Thanks!!!

cotestatnt, 2023-03-27:
Hi @pablopeu 
Maybe something gets broken with latest updates.
I will check as soon as possible

cotestatnt, 2023-03-28:
Good morning @pablopeu 
I've tested the sketch ButtonQuery.ino and I have the same issue.
It's not clear to me why ESP8266 do this (ESP32 work flawlessy), but it seems that the restart instruction inside the update handler don't work as it should.

Anyway, if you move the restart handling in the main loop using a bool flag in order to check when perform restart, it works.
Try this:
```C++
/*
  Name:	       OTA_buttonQuery.ino
  Created:     29/03/2021
  Author:       Tolentino Cotesta <cotestatnt@yahoo.com>
  Description: an example that check for incoming messages
              and install rom update remotely.
*/
#include <AsyncTelegram2.h>
#ifndef LED_BUILTIN
#define LED_BUILTIN 2
#endif

// Timezone definition
#define MYTZ "CET-1CEST,M3.5.0,M10.5.0/3"
#include <time.h>

#ifdef ESP8266
#include <ESP8266WiFi.h>
  #include <ESP8266HTTPClient.h>
  #include <ESP8266httpUpdate.h>
  #define UPDATER ESPhttpUpdate
  Session   session;
  X509List  certificate(telegram_cert);
#elif defined(ESP32)
  #include <HTTPClient.h>
  #include <HTTPUpdate.h>
  #define UPDATER httpUpdate
#endif

WiFiClientSecure client;
AsyncTelegram2 myBot(client);
const char* ssid  =  "xxxxxx";     // SSID WiFi network
const char* pass  =  "xxxxxxx";     // Password  WiFi network
const char* token =  "xxxxxxxxxxxxx:xxxxxxxxxxxxx";  // Telegram token

// Check the userid with the help of bot @JsonDumpBot or @getidsbot (work also with groups)
// https://t.me/JsonDumpBot  or  https://t.me/getidsbot
int64_t userid = 123456789;

#define CANCEL  "CANCEL"
#define CONFIRM "FLASH_FW"

const char* firmware_version = __TIME__;
bool reset = false;

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  delay(500);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(100);
  }

#ifdef ESP8266
  // Sync time with NTP, to check properly Telegram certificate
  configTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
  //Set ceritficate, session and some other base client properies
  client.setSession(&session);
[... 131 lines cut ...]
  Serial.println("Update done!");
  client.stop();

  switch (ret) {
    case HTTP_UPDATE_FAILED:
      report = "HTTP_UPDATE_FAILED Error (";
      report += UPDATER.getLastError();
      report += "): ";
      report += UPDATER.getLastErrorString();
      myBot.sendMessage(msg, report.c_str());
      break;

    case HTTP_UPDATE_NO_UPDATES:
      myBot.sendMessage(msg, "HTTP_UPDATE_NO_UPDATES");
      break;

    case HTTP_UPDATE_OK:
      myBot.sendMessage(msg, "UPDATE OK.\nRestarting in few seconds...");
      reset = true;
      break;
    default:
      break;
  }
}
```

closed 2023-07-25 by cotestatnt