cotestatnt/AsyncTelegram2 #102: restart loop, message index not updated?

sanderkob, 2023-03-27:
In my sketches I occasionally want to restart by issuing a "/restart" command via Telegram.
I scan the incoming Telegram message for the text "/restart" and if present issue an ESP.restart() call.
In earlier telegram bot libs this worked well.
But in asynctelegram2 the program (bot) keeps restarting. After a restart, the first new message is in fact the old message, that contains the "/restart"  text, so a new ESP.restart()  call is issued.
This becomes an endless loop.
I tested with the echoBot example with debug log enabled, running on an ESP32, version 2.0.7
The same happened on an ESP8266 board, version 3.1.1
Details of program and output are in https://gist.github.com/sanderkob/09719d4ffa42f868949298cef7735ef6

cotestatnt, 2023-03-28:
Hi @sanderkob 
This is as the issue https://github.com/cotestatnt/AsyncTelegram2/issues/101
I'm working on it.

cotestatnt, 2023-03-28:
I looked at your sketch and I have to correct what I wrote before: it's not the same problem.

To avoid the bootloop you need to be sure that bot is synced with Telegram server.
To do this, I added the `noNewMessage()` method which you can call as long as it returns a false value, at which point you can successfully restart.

```C++
    if (msg.text == "/restart") {
      while (!myBot.noNewMessage()) {
        Serial.print(".");
        delay(50);
      }
      ESP.restart();
    }
```

closed 2023-03-28 by sanderkob