cotestatnt/AsyncTelegram2 #103: Buffersize wificlientsecure

sanderkob, 2023-03-28:
I noticed in the examples that the client buffersizes are set to 1k with setBuffersizes(1024,1024).
As far as I understand the safe buffersizes are (16k, 512). Only when the server supports MFLN (TLS 1.2), the negotiated buffer sizes can be safely reduced.
The buffersize is relevant for ESP8266 with its limited RAM.
Do you recommend to generally reduce buffersizes with the Telegram server?

cotestatnt, 2023-03-29:
Hi @sanderkob and thank you for your interest in my library.

Luckily the api.telegram.org server supports the MFLN option (TLSv1.3) so it is possible to reduce the buffer size and save some memory.

Thinking about it, in all probability  the rx buffer is set a little too low and the tx buffer too high (which by default would be 512).
I have to be honest: I haven't used ESP8266s for a long time and therefore I haven't paid much attention to the examples for the part of the code that concerns these MCUs.
if you have more experience or any kind of advice, please share! It will be really appreciated

sanderkob, 2023-04-28:
I am now running a similar async client on ESP8266 based on a state machine: TelegramBotClient ([https://github.com/schlingensiepen/TelegramBotClient](https://github.com/schlingensiepen/TelegramBotClient)). 
I made some changes, I will prepare a fork one of these days.
With TelegramBotClient I get similar response times as with your client, 600-800 ms, 
Only problem I had was that my programme used so much memory.
Your use of setBufferSizes() made me check. My TelegramBotClient uses the default values (16384,512) of WiFiClientBearSSL.
With setBufferSizes(4096,512) the programme runs smooth and leaves me some memory.
I will stick with my TelegramBotClient, but thanks

closed 2023-06-07 by cotestatnt