cotestatnt/AsyncTelegram2 #106: getMaxAllocHeap in send message method causes issues

MicDG, 2023-04-26:
Hello, as somebody else reported in issue #83 I encountered some problems when sending messages due to this lines in the sendMessage(...) method (I am using an ESP32):

```
#if defined(ESP8266)
            DynamicJsonDocument doc(ESP.getMaxFreeBlockSize() - BUFFER_MEDIUM);
#elif defined(ESP32)
            DynamicJsonDocument doc(ESP.getMaxAllocHeap);
#else
            DynamicJsonDocument doc(BUFFER_BIG);
#endif
```

The keyboard I am generating needs a DynamicJsonDocument with 768 bytes of capacity to deserialize it correctly (checked this with the ArduinoJson Assistant), but it fails with a NoMemory error even though getMaxAllocHeap() reports over 49 kB of allocatable heap. In fact the document isn't created at all (check with capacity() ). Changing the code as such makes it work:

```
#elif defined(ESP32)
            DynamicJsonDocument doc(BUFFER_BIG);
```

Maybe the code should be adjusted to use a buffer size chosen via defines?

cotestatnt, 2023-04-27:
Hi @MicDG 
I've removed those lines along all the source files.

In order to set a custom json buffer size, you can use the metod [setJsonBufferSize()](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.h#L432)

MicDG, 2023-04-28:
Thank you very much for the quick fix!

closed 2023-05-02 by cotestatnt