cotestatnt/AsyncTelegram2 #107: Adding a delete message method

MicDG, 2023-04-28:
Hello, just wondering if something like this could be added to delete a message:

```
bool AsyncTelegram2::deleteMessage(const TBMessage &msg)
{
    char payload[BUFFER_SMALL];
    snprintf(payload, BUFFER_SMALL,
        "{\"chat_id\":%lld,\"message_id\":%" INT32 "}",
        msg.chatId, msg.messageID);

    const bool result = sendCommand("deleteMessage", payload);
    log_debug("%s", payload);
    return result;
}
```

Thank you very much!

cotestatnt, 2023-05-02:
Hi @MicDG 
In order to delete a message you need two things as I assume you have already checked:
- the chat ID (user or groups)
- the message ID

With that method you will be able to delete just in case you are replying to a received message you.
So I think it would be better defining something more "general" and then ovverride the method:

```C++
bool AsyncTelegram2::deleteMessage(const char* chatId, const char* messageId)
{
    char payload[BUFFER_SMALL];
    snprintf(payload, BUFFER_SMALL,
        "{\"chat_id\":%lld,\"message_id\":%" INT32 "}",
         chatId, messageId);

    const bool result = sendCommand("deleteMessage", payload);
    log_debug("%s", payload);
    return result;
}

bool AsyncTelegram2::deleteMessage(const TBMessage &msg) 
{
    return deleteMessage(msg.chatId, msg.messageID);
}
```

MicDG, 2023-05-04:
That makes sense, should I make a pull request to add it to the library?

cotestatnt, 2023-05-05:
Yes, if you can.

Otherwise I will add in the next release.

closed 2023-06-07 by cotestatnt