cotestatnt/AsyncTelegram2 #108: "<" seems to break something?

wardom, 2023-05-01:
Hello Tolentino, love your work!

I'm new to this, so I might be doing something wrong and apologise if so.

While tinkering, I've found that by adding "<" to an outgoing message the bot doesn't send the message. I've tried using the example codes you've provided but none of them get sent if the message includes "<".

I also tried the "echoBot" example provided. It works flawlessly but if the message includes "<" the bot will not echo the message.
[image]
This was done using the echoBot sketch. Other symbols don't seem to cause any problems.

I hope I explained well enough, please let me know if you need anymore information.
Thank you for your time.

sanderkob, 2023-05-01:
See
https://stackoverflow.com/questions/40626896/telegram-does-not-escape-some-markdown-characters
where it is mentioned that
"<" will be read as tag start and probably unrecognized.

cotestatnt, 2023-05-02:
Hi @sanderkob 
The default formatting options used by AsyncTelegram2 library is HTML
The char '<' it's an HTML start TAG so probally the server parser expect the remaining code and this cause error.

Try [switching to the MarkdownV2](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/formatStyle/formatStyle.ino) formatting style.

Check the example or simply add this line:
`myBot.setFormattingStyle(AsyncTelegram2::FormatStyle::HTML /* MARKDOWN */);`

You can try also to encode your message text using [entity name or entity number](https://www.w3schools.com/html/html_entities.asp):
<table class="ws-table-all notranslate">
    <tbody><tr>
      <th>Result</th>
      <th>Description</th>
      <th>Entity Name</th>
      <th>Entity Number</th>
    </tr>
    <tr>
      <td style="height: 29px"></td>
      <td style="height: 29px">non-breaking space</td>
      <td style="height: 29px">&amp;nbsp;</td>
      <td style="height: 29px">&amp;#160;</td>    
    </tr>
    <tr>
      <td>&lt;</td>
      <td>less than</td>
      <td>&amp;lt;</td>
      <td>&amp;#60;</td>
    </tr>
    <tr>
      <td>&gt;</td>
      <td>greater than</td>
      <td>&amp;gt;</td>
      <td>&amp;#62;</td>
    </tr>
    <tr>
      <td>&amp;</td>
      <td>ampersand</td>
      <td>&amp;amp;</td>
      <td>&amp;#38;</td>
    </tr>
    <tr>
      <td>"</td>
      <td>double quotation mark </td>
      <td>&amp;quot;</td>
      <td>&amp;#34;</td>
    </tr>
    <tr>
      <td>'</td>
      <td>single quotation mark (apostrophe) </td>
      <td>&amp;apos;</td>
      <td>&amp;#39;</td>
    </tr>
    <tr>
      <td>¢</td>
      <td>cent</td>
      <td>&amp;cent;</td>
      <td>&amp;#162;</td>
    </tr>
    <tr>
      <td>£</td>
      <td>pound</td>
      <td>&amp;pound;</td>
      <td>&amp;#163;</td>
    </tr>
    <tr>
      <td>¥</td>
      <td>yen</td>
      <td>&amp;yen;</td>
      <td>&amp;#165;</td>
    </tr>
    <tr>
      <td>€</td>
      <td>euro</td>
      <td>&amp;euro;</td>
      <td>&amp;#8364;</td>
    </tr>
    <tr>
      <td>©</td>
      <td>copyright</td>
      <td>&amp;copy;</td>
      <td>&amp;#169;</td>
    </tr>
    <tr>
      <td>®</td>
      <td>registered trademark</td>
      <td>&amp;reg;</td>
      <td>&amp;#174;</td>
    </tr>
    </tbody></table>

wardom, 2023-05-03:
Hello, thank you for the replies

Added the line:
`myBot.setFormattingStyle(AsyncTelegram2::FormatStyle::HTML /* MARKDOWN */);`
Now the bot is able to display the char "<" when using entity names or entity numbers.

Appreciate the help and looking forward to more of your work!

closed 2023-05-03 by wardom