cotestatnt/AsyncTelegram2 #112: esp32-cam no messages with long send interval

tiffass, 2023-07-02:
Hi all!
There is an ESP32-cam module and the need to send messages to telegrams at a certain interval. It would seem nothing complicated. Start the timer, send messages when the specified interval is reached and rejoice. But that was not the case... If I set the sending interval>15 minutes, then for some reason the messages do not arrive, while the messages are displayed in the port monitor. If <=15 then everything is OK. Who can tell what is the problem? Is this a feature of this module, library, or am I doing something wrong in the code? Here is a minimal example of my code:

`#include "soc/soc.h"
#include "soc/rtc_cntl_reg.h"
#include <time.h>
#include <AsyncTelegram2.h>

#ifdef ESP8266
#include <ESP8266WiFi.h>
BearSSL::WiFiClientSecure client;
BearSSL::Session   session;
BearSSL::X509List  certificate(telegram_cert);

#elif defined(ESP32)
#include <WiFi.h>
#include <WiFiClient.h>
#if USE_CLIENTSSL
#include <SSLClient.h>
#include "tg_certificate.h"
WiFiClient base_client;
SSLClient client(base_client, TAs, (size_t)TAs_NUM, A0, 1, SSLClient::SSL_ERROR);
#else
#include <WiFiClientSecure.h>
WiFiClientSecure client;
#endif
#endif

AsyncTelegram2 bot(client);

const char* ssid = "ssid";
const char* pass = "pass";

#define BOT_TOKEN "token"
#define CHAT_ID 11111111

#define MYTZ "EET-2EEST,M3.5.0/3,M10.5.0/4"

void setTimezone(String timezone) {
  Serial.printf("  Setting Timezone to %s\n", timezone.c_str());
  setenv("TZ", timezone.c_str(), 1); //  Now adjust the TZ.  Clock settings are adjusted to show the new local time
  tzset();
}

unsigned long send_time;
unsigned long last_msg = 30*60*1000;
bool flag = 0;

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);
   Serial.begin(115200);
  WiFi.begin(ssid, pass);
  byte tries = 10;
  while (--tries && WiFi.status() != WL_CONNECTED) {
    delay(500);
  }

#if USE_CLIENTSSL == false
  client.setCACert(telegram_cert);
#endif

  configTzTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");

  struct tm timeinfo;
  if (getLocalTime(&timeinfo)) {
    setTimezone(MYTZ);
  }

  bot.setUpdateTime(1000);
  bot.setTelegramToken(BOT_TOKEN);
  bot.begin() ? Serial.println("OK") : Serial.println("NOK");
bot.sendTo(CHAT_ID, "Start");
 
}

void loop() {

  TBMessage msg;

  if (millis() - send_time >= last_msg) {
    
    send_time = millis();
    flag = 0;
    Serial.println("SEND");
    if (!flag && WiFi.status() == WL_CONNECTED) {
      bot.sendTo(CHAT_ID, "timeGet");
      Serial.println("SEND bot");
      flag = 1;
    }
  }

}`

cotestatnt, 2023-07-25:
Hi @tiffass 

This is the same issue as https://github.com/cotestatnt/AsyncTelegram2/issues/81

You need to call getNewMessage() periodically or at least shortly before sending a new message to synchronize the bot with the server (and eventually re-establish the connection)

tiffass, 2023-07-27:
Hi @cotestatnt 
Yes, I already saw that I missed this function and then added it, but it did not solve the problem

ardumiguel, 2023-09-04:
Im facing the same issue too!
Althought with a higher time span, like one day or so.

my code already implements `getNewMessage()`:

` 
if (WiFi.status() == WL_CONNECTED) {
    if (myBot.getNewMessage(msg)) {

    }
}
`