cotestatnt/AsyncTelegram2 #114: prevent restart loop

sanderkob, 2023-08-02:
I have implemented a restart command in my bot. When i issue the command `/restart`,  in my code this is interpreted and an `ESP.restart() `command is issued. <br>Apparently the message queue is not cleared because after restart the first message received is `/restart` and again an `ESP.restart()` command is issued.<br> This results in an endless loop. How can I programmatically clear the message queue before issuing the `ESP.restart()` command?<br> I tried <br>
```
getNewMessage(msg);
ESP.restart();
```
 but that does not work.

cotestatnt, 2023-08-02:
Hi @sanderkob 
This is the same issue as https://github.com/cotestatnt/AsyncTelegram2/issues/102

sanderkob, 2023-08-02:
Correct, my oversight.

closed 2023-08-02 by sanderkob