cotestatnt/AsyncTelegram2 #115: esp32-cam getting photo

tiffass, 2023-08-06:
Using the ESP32-cam.ino example. Why do I need to send the /takePhoto command twice to get the current photo? At the first sending, a previously taken photo arrives, and only after the second sending of the command I already receive a new photo. Is this a bug in the code or a feature of telegrams?

cotestatnt, 2023-08-11:
Hi @tiffass 
I've just uploaded a revisited and working example for ESP32-CAM

tiffass, 2023-08-12:
No, it doesn't work either.
First, there is a compilation error:
**sketch_aug12b:72:1: error: 'camera_config_t' has no non-static data member named 'sccb_i2c_port'
 };
 ^
exit status 1
'camera_config_t' has no non-static data member named 'sccb_i2c_port'**

If I comment the line
**.sccb_i2c_port = 1**
compilation succeeds.
Secondly, after loading the code and sending a command to get a photo from the module, the previous photo still comes first, and only when the request is sent the second time, I get the current photo. Here is a screenshot from the Telegram screen 
[image]
(I replaced /takePhoto with /on)

cotestatnt, 2023-08-12:
Which version of [ESP32 Arduino core](https://github.com/espressif/arduino-esp32) are you using?

tiffass, 2023-08-12:
2.0.2