cotestatnt/AsyncTelegram2 #116: HC-SR501 and ESP32-CAM does not work simultaneously with SD card

tiffass, 2023-08-06:
Example ESP32-CAM-PIR.ino does not work with motion sensor HC-SR501. When the sensor is connected at the initialization stage, the error **"SD Mount Failed"** occurs. And accordingly further the code works incorrectly. More precisely, motion is detected, but photos are not sent to telegrams. The error **"Failed to open file in writing mode"** falls out in the serial port. If the sensor is turned off during the start of the module, then everything is OK. At rest, the output of the sensor is 0, and as I understand it, this prevents the SD card from being initialized. Can this be fixed somehow?

cotestatnt, 2023-08-11:
It would appear that you are using pins used by the SD module.
Have you tried connecting the HC-SR501 module differently?

tiffass, 2023-08-12:
Yes, I connected the HC-SR501 to the pin, which is registered in the example
**#define PIR_PIN GPIO_NUM_13**
and the same pin is used in the ESP32-CAM for the SD card.
And for some reason, this pin 13 is used in all the examples found on the network to connect a motion sensor.
From the description of the module, it can be seen that there are pins 0,1,3,16 available. Pin 0 in this case is also not desirable to use, because. it is involved when the module is loaded. 1.3 is Serial, you can use it, but then it will not be possible to fully use the serial port monitor. Only gpio16 remains?
[image]

cotestatnt, 2023-08-21:
The example provided with library, use the onboard SD card reader in **1-bit mode**.
In this way only GPIO02 is used to read and write data to the SD card and the other pins are free to use.

```
if (!FILESYSTEM.begin("/sdcard", true, true)) {
....
```

The problem is due to how the pin has been configured to easily test the sketch: the input is tested as a simple button that closes towards GND, but normally the PIR sensors work in reverse generating a positive pulse every time an object is detected.

If you set the PIN correctly and change the related if code block like this it should work as expected.

```
   // PIR Motion Sensor setup
  pinMode(PIR_PIN, INPUT);      
  .....
  .....

    // PIR motion detected 
    // could be on interrupt, but PIR sensor usually keep signal HIGH for enough time
    if(digitalRead(PIR_PIN)) {   
    .....
    .....

```

tiffass, 2023-08-26:
@cotestatnt 
No, I tried before creating the issue and it doesn't work either. If I understand correctly, then according to this https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html#pull-up-conflicts-on-gpio13 on Pin 13 to initialize the card should be logical 1. And at the output of the sensor in the absence of movement - logical 0. Therefore, when starting esp32-cam on pin 13 - 0 and the SD card does not initialize

[image]

closed 2025-04-05 by cotestatnt