cotestatnt/AsyncTelegram2 #117: Async web request conflicts with Async telefgram2.

miguelfernando55, 2023-08-09:
I am trying to open a webpage using server.on() at the same time using bot asynctelegram2. But the web page does not open. The bot still runs.

cotestatnt, 2023-08-11:
Can you provide a sketch which can reproduce your issue? 
I've just tried to serve a simple web page and it works.

miguelfernando55, 2023-08-12:
Sorry, I cannot. It does not belong to me. 
Yes, with only a few lines of code it works. I also did only with a few lines before going further, but for a 40kB html css js it creates a err_invalid_chunked_encoding. 
I thought the cause was something related to the quality of the content or chrome but when I comment the lines. myBot.begin and getNewMessages it works fine. 
I also tries to bundle the html js css but the code is delivery partially to the browser.

cotestatnt, 2023-08-21:
If you are using ESPAsynWebServer library could be something like related to this issue

cotestatnt, 2023-08-21:
If you are using ESPAsynWebServer library could be something like related [to this issue](https://github.com/me-no-dev/ESPAsyncWebServer/issues/1167) compounded by the additional resources required for the Telegram WiFi client.

With the suggested pull requests, I have greatly improved the stability of some of my projects

miguelfernando55, 2023-08-21:
It appears that to create a server and client at the same time is not possible. I do not know why.

closed 2025-04-05 by cotestatnt