cotestatnt/AsyncTelegram2 #118: AsyncTelegram2 hangs the uC, when "unable to connect to server"

miguelfernando55, 2023-08-09:
When it is not connected to the server, the uC hangs. It also should be async.

cotestatnt, 2023-08-09:
Hi @miguelfernando55 .
Which microcontroller is it?

miguelfernando55, 2023-08-09:
esp32-s2-saola.
Here what it is happenning 
[image]
When it is unable, it delays the normal functionality of the rest of the system.
As you may check in the image, I have an interval to send new info. When it happens "unable ... ", it is much longer

cotestatnt, 2023-08-09:
Thanks for your feedback, I will try to check after 20th August because I am going on vacation from next weekend

miguelfernando55, 2023-08-09:
Ok. Nice holidays. Hope you may upgrade the functionality.

miguelfernando55, 2023-08-10:
Is it able to use a flag that warns when the "unable to connect to server" happens?

cotestatnt, 2023-08-21:
Hi @miguelfernando55 
Regarding your issue, it is not clear to me if the failed connection to the Telegram server happens even if the esp32 is correctly connected to the internet.

If you don't have an active internet connection, you should avoid calling the `getNewMessage()` method (and more generally the other library methods) because an [attempt to connect to the server](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.cpp#L29) will still be made and this step is blocking (you can set a timeout eventually).

Checking the internet connection cannot be "internal" to the library because it is strictly linked to the hardware interface (WiFi, Ethernet, GPRS etctec).

miguelfernando55, 2023-08-21:
You are right. The esp32 may be connected to the router, but the router lost internet connection. It was because of this situation that the "unable to ...." happenned.
I am blocking and not able to design a way to avoid this situation.

miguelfernando55, 2023-08-22:
Are you able to provide me a status variable when the connection is hanging?