cotestatnt/AsyncTelegram2 #125: Example EchoBot takes up a lot of memory in ESP32

SergGrn, 2023-09-10:
Hello, dear Tolentino.
This is not a problem or a bug, but rather a question, sorry if I posted it incorrectly.
I'm using the echoBot example, on an ESP32, CLIENTSSL connection.
On Arduino IDE 2.2.0. AsyncTelegram2 version 2.2.2.
I can't figure out why this example sketch takes up 59% of memory (773949 byte).
The fact is that earlier I used the version of the AsyncTelegram2 2.0.8 library, while using the Arduino IDE 1.8.19. And with the same controller, the sketch, which is much larger and has many other functions besides Telegram, takes up about 48% of the memory.
What could be the reason?
Thank you!

cotestatnt, 2023-09-11:
Could you provide more detail for your request?
Which version of ESP32 Arduino Core?
Which board did you select?

You should then compare by including as few external libraries as possible. SSLClient.h is a library that makes heavy use of SRAM for example, try without.

Anyway, I also tried with these configurations:
ESP32-S3 development board, ESP32 Core 2.0.9, AysncTelegram 2.2.3 and AsyncTelegram 2.0.8 and I more or less always have the same values.

However, rather than the message provided by the compiler, you should shift your attention to the amount of SRAM that remains free during program execution.
Those percentages are an estimate based on the variable declarations, but for those that are dynamically allocated, the compiler doesn't know in advance how much space they will take up in memory.
For example, in the latest releases, I changed the declaration of some variables, making them static. In this way, the compiler includes them in the calculation of the resources used with more precision.

SergGrn, 2023-09-11:
Version of ESP32 Arduino:
on IDE 1.8.19 - version core 2.0.9;
on IDE 2.2.0 - version core 2.0.2.
In both cases, the "ESP32 Dev Module" board is selected.

SergGrn, 2023-09-11:
Now I have updated the ESP32 core version to 2.0.11 in Arduino 2.2.0.
This is what the IDE showed as a result of compilation:
"The sketch uses 846257 bytes (64%) of device memory. A total of 1310720 bytes are available.
Global variables use 52796 bytes (16%) of dynamic memory, leaving 274884 bytes for local variables. Maximum: 327680 bytes."

Here's my entire sketch:
`
#include <WiFi.h>
#include <WiFiClient.h>
#include <AsyncTelegram2.h>

#include <time.h>
#define MYTZ "EET-2EEST,M3.5.0/3,M10.5.0/4"

#include <SSLClient.h>  
#include "tg_certificate.h"
WiFiClient base_client;
SSLClient client(base_client, TAs, (size_t)TAs_NUM, A0, 1, SSLClient::SSL_ERROR);

AsyncTelegram2 myBot(client);

const char* ssid  =  "********";     // SSID WiFi network
const char* pass  =  "********";     // Password  WiFi network
const char* token =  "********";  // Telegram token

int64_t userid = ********;  

void setup() {
  pinMode(2, OUTPUT);
 
  Serial.begin(9600);
  Serial.println("\nStarting TelegramBot...");

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  delay(500);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(500);
  }
 

  configTzTime(MYTZ, "time.google.com", "time.windows.com", "pool.ntp.org");
 

  myBot.setUpdateTime(2000);
  myBot.setTelegramToken(token);

  myBot.begin();

}

void loop() {
  
  static uint32_t ledTime = millis();
  if (millis() - ledTime > 500) {
    ledTime = millis();
    digitalWrite(2, !digitalRead(2));
  }

  TBMessage msg;

  if (myBot.getNewMessage(msg)) {    
    String message;
    message += "Message from @";
    message += myBot.getBotName();
    message += ":\n";
    message += msg.text;
    myBot.sendMessage(msg, message);
  }
}`