cotestatnt/AsyncTelegram2 #130: removeReplyKeyboard not working

panditjs, 2023-10-31:
replaykeyboard could not remove please help

cotestatnt, 2023-11-07:
Hi @panditjs 
This bug should be fixed with latest release.

panditjs, 2023-11-07:
Ok i downgrade to previous version 2.2.1 and it work again perfecly thanks

panditjs, 2023-11-07:
I will try the new version 2.2.3 thanks

NamDuong9999, 2024-01-09:
................
Test Telegram connection... OK

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40201c47 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000002a depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff860 end: 3fffffd0 offset: 0150
3ffff9b0:  00004cc8 00010000 3ffe8664 40100816  
3ffff9c0:  3ffffb60 00000000 3ffe8cd7 40202fed  
3ffff9d0:  00004b30 00000966 3ffe8664 40100816  
3ffff9e0:  00000006 00000020 00000404 3ffe864c  
3ffff9f0:  00000000 00000020 3fff46b4 00000000  
3ffffa00:  3ffe8cd7 3fff46c4 3ffffb60 40203078  
3ffffa10:  00000000 00000020 3fff0044 40100b64  
3ffffa20:  3ffe8cd7 3ffffb60 3ffffb60 00000000  
3ffffa30:  3ffe8da6 3fff46c4 3ffffb60 40205cdf  
3ffffa40:  3ffea1e0 00000026 00000002 007cd7b2  
3ffffa50:  3ffea204 2c9f0300 4000050c 3fffc278  
3ffffa60:  401030a0 3fffc200 00000022 2c9f0300  
3ffffa70:  40222277 00000030 00000010 ffffffff  
3ffffa80:  40208daa 3fff28ec 0000012c 00000000  
3ffffa90:  00004bc6 00000000 00000000 fffffffe  
3ffffaa0:  00000000 3fffc6fc 00000000 3fff2b94  
3ffffab0:  3fff28ec 00001f11 00000001 00000030  
3ffffac0:  00000000 00000000 00000000 fffffffe  
3ffffad0:  ffffffff 3fffc6fc 00000001 3ffe8660  
3ffffae0:  000002db 00001f11 00000001 00000030  
3ffffaf0:  000002db 00001f11 00000001 00000030  
3ffffb00:  00000000 3fffc6fc 00000000 3fff0664  
3ffffb10:  00000000 00000000 3fff46ec 00000030  
3ffffb20:  00000000 00001f11 49ba5e35 0023ebfd  
3ffffb30:  3fff4261 00000000 00000000 00000002  
3ffffb40:  00000001 00000000 3fff2b94 40208eea  
3ffffb50:  00000000 00001f11 63d70a3d 0023ec1b  
3ffffb60:  3ffe864c 00000000 00000000 00000000  
3ffffb70:  00000000 00000000 3fff0664 40209e1c  
3ffffb80:  00000000 00000000 3ffe8660 3ffffb6c  
3ffffb90:  00040000 0000ffff 00010000 00000001  
3ffffba0:  00001f20 00000430 00000800 40208dbf  
3ffffbb0:  000002db 0000012c 3fff2fac 3fff46ec  
3ffffbc0:  00000000 00000000 3fff0664 3fff46ec  
3ffffbd0:  00004b68 0000096d 3ffe8664 40100816  
3ffffbe0:  00000400 00000001 3ffffc20 3ffe8d62  
3ffffbf0:  00000000 00000000 0000001f 401001ec  
3ffffc00:  3fff2c14 00000001 3fffc228 40105c8d  
3ffffc10:  4000050c 00000020 3fff46b4 40100b64  
3ffffc20:  402124bd 00000030 00000010 ffffffff  
3ffffc30:  402016f5 3ffffe40 00000001 3fffff40  
3ffffc40:  00000000 00000000 00000000 fffffffe  
3ffffc50:  00000000 3fffc6fc 00000000 3ffef000  
3ffffc60:  3ffeeff8 3fffdad0 3ffef394 00000030  
3ffffc70:  3ffe88ea 3ffe83d0 3ffffd30 4021ab05  
3ffffc80:  00000000 00000012 00000000 00000000  
3ffffc90:  00000012 0000003c 3f302073 3ffe864c  
3ffffca0:  00000000 00000020 3fff46b4 40100b64  
3ffffcb0:  00000000 3ffffd5c 3ffffd5c 40212638  
3ffffcc0:  3ffffd48 3ffffd10 3fff2bcc 40100b64  
3ffffcd0:  3fff4704 3ffe864c 00000000 00000001  
3ffffce0:  3ffffe40 3ffef004 00000010 40205049  
3ffffcf0:  00000000 0000209a b74bc6a7 0025b2e3  
3ffffd00:  00000025 3fff2042 3ffffe60 3ffef394  
3ffffd10:  3fffdad0 3ffef004 3ffef000 40201702  
3ffffd20:  00000000 3ffe83d0 00000080 4021624d  
3ffffd30:  3fffff3c 3ffef394 00000043 3ffffd1c  
3ffffd40:  00040000 0000ffff 00020000 40209ee8  
3ffffd50:  3ffe8620 00000000 00000800 00000000  
3ffffd60:  00000000 00000000 00001def 3ffef394  
3ffffd70:  3fffff00 00000040 00000020 40100b98  
3ffffd80:  3fffff00 00000040 3fffff8c 4020d30b  
3ffffd90:  00003048 00000609 3ffe8664 40100816  
3ffffda0:  0000000c fffffffc 00000000 3ffef394  
3ffffdb0:  3ffef084 00000020 3ffffe38 402011b1  
3ffffdc0:  3ffef084 3ffef1f8 3ffffe00 402011b1  
3ffffdd0:  3ffef084 3ffef1f8 3ffef004 402013cc  
3ffffde0:  feefeffe 00000001 feefeffe feefeffe  
3ffffdf0:  6ac657a7 00000000 feefeffe feefeffe  
3ffffe00:  00000005 00000000 00000020 401001ec  
3ffffe10:  00000000 feefeffe 00000005 4010215c  
3ffffe20:  3ffe9965 401053f3 3ffed378 feefeffe  
3ffffe30:  40102ce7 3ffed378 feefeffe feefeffe  
3ffffe40:  00000000 00000001 3ffedcc0 40102ec4  
3ffffe50:  3ffea1f8 00000000 00000000 00000000  
3ffffe60:  00007fff 007f5192 00000000 00000000  
3ffffe70:  00000000 7fffffff 00000000 00000000  
3ffffe80:  00000001 00000000 00000000 00000000  
3ffffe90:  00000000 00000000 00000000 007f5192  
3ffffea0:  00000000 00000000 00000000 3fffc278  
3ffffeb0:  00000000 00000000 00000022 00000000  
3ffffec0:  00000000 00000030 00000000 ffffffff  
3ffffed0:  4020e7b0 40100e2c 4020e7a4 00000000  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  ffffffff 00000000 00000000 3fffdab0  
3fffff00:  00000000 00000000 3ffef394 00000000  
3fffff10:  5f6d654d 20746f62 00000000 2f0a656e  
3fffff20:  00000000 00000000 6d6f6320 00000000  
3fffff30:  00000000 69616c61 00000000 feefef00  
3fffff40:  feefeffe feefeffe 00000000 feefeffe  
3fffff50:  00000000 00000000 feefeffe 00000000  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  00000000 feefeffe 00000000 00000000  
3fffff90:  003c003f 00000000 feefeffe 3ffef394  
3fffffa0:  3fffdad0 00000000 3ffef368 3ffef394  
3fffffb0:  3fffdad0 00000000 3ffef368 4020e7c4  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100e31  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0006dac0
~ld
⸮⸮⸮⸮g⸮⸮s⸮⸮'|⸮$⸮l`c⸮⸮|r⸮l⸮o⸮⸮g⸮