cotestatnt/AsyncTelegram2 #133: Guru Meditation Error: Core 1 panic’ed since ArduinoJson release 7.0.0

ardumiguel, 2024-01-14:
Pretty much what the title says. ESP32 hangs with the mentioned error upon receiving a message if I update Arduino Json to anything past versino 7.0.0.

TechWellLiving, 2024-01-14:
Similar error seen with ESP32 S3 and ESP EYE, as soon as i call   **myBot.getNewMessage()**, it throws this error;

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d3f52  PS      : 0x00060f30  A0      : 0x800d5376  A1      : 0x3ffcfb50  
A2      : 0x00000002  A3      : 0x3ffcfd58  A4      : 0x3ffdf374  A5      : 0x0000000a  
A6      : 0x3ffdf384  A7      : 0x00000001  A8      : 0x0000ffff  A9      : 0x3ffcfb30  
A10     : 0x00000020  A11     : 0x00000020  A12     : 0x00001800  A13     : 0x3ffdf374  
A14     : 0x3ffb657c  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002a  LBEG    : 0x4008bd15  LEND    : 0x4008bd25  LCOUNT  : 0xfffffffe  

Backtrace: 0x400d3f4f:0x3ffcfb50 0x400d5373:0x3ffcfb80 0x400d53f1:0x3ffcfbb0 0x400d56a2:0x3ffcfbe0 0x400d7de5:0x3ffcfc00 0x400d3cad:0x3ffcff40 0x400e1b31:0x3ffd00c0

T-helper, 2024-01-15:
Faced with it too. 
i spent a sleepless night trying to catch that strange error. Thanks for solution with arduino json

cotestatnt, 2024-01-15:
Thanks everyone for reporting. 
I'm working on it, as soon as the bug is found I will probably publish a new a major release because I'm also migrating to the new APIs of the ArduinoJson library (but maintaining backward compatibility)

cotestatnt, 2024-01-16:
Hi everyone. 
I just published a new release that includes full support for the new major version 7 of ArduinoJson and compatibility for version 6.

closed 2024-01-16 by ardumiguel