cotestatnt/AsyncTelegram2 #135: No issue - feature request for ESP32-CAM

FBMinis, 2024-01-28:
Hi Tolentino,

I have been using your library for almost 3 years, thank you for sharing your work. I modified one of the examples so my battery powered ESP32CAMs turn ON and send a photo, either using PIR as a trigger or a DS3231 to define the interval of time. One of them as sent me more than 3000 photos since I last charge the battery, the connection to Telegram works perfectly.

I would like the photo to include a caption, which would be a variable with a timestamp, previously stored in LittleFS. How do you add a caption to a photo?

The following code is what I use to send a photo to my Telegram account when I power ON the device:

```
//ESP32-CAM sends a photo to Telegram when power is applied

/*----------------------------------------------------------------------------------------------------------------------------
  Used: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi
  Not used: E:\arduino-1.8.19\libraries\WiFi
  Using library WiFi at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi
  Using library AsyncTelegram2 at version 2.2.1 in folder: E:\arduino-1.8.19\portable\sketchbook\libraries\AsyncTelegram2
  Using library ArduinoJson at version 6.21.3 in folder: E:\arduino-1.8.19\portable\sketchbook\libraries\ArduinoJson
  Using library FS at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\FS
  Using library WiFiClientSecure at version 2.0.0 in folder: E:\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFiClientSecure
  "E:\\arduino-1.8.19\\portable\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-2021r2-patch5-8.4.0/bin/xtensa-esp32-elf-size" -A "C:\\Users\\user\\AppData\\Local\\Temp\\arduino_build_583398/sendPhotoTelegramStartup.ino.elf"
  Sketch uses 1014089 bytes (32%) of program storage space. Maximum is 3145728 bytes.
  Global variables use 49984 bytes (15%) of dynamic memory, leaving 277696 bytes for local variables. Maximum is 327680 bytes.
  ----------------------------------------------------------------------------------------------------------------------------*/

#include <WiFi.h>

#include <AsyncTelegram2.h> //https://github.com/cotestatnt/AsyncTelegram2/tree/2.3.0

#include "esp_camera.h"
#include "soc/soc.h"           // Brownout error fix
#include "soc/rtc_cntl_reg.h"  // Brownout error fix

#include <WiFiClientSecure.h>
WiFiClientSecure secure_client;

const char* ssid = "blabla";                                        // SSID WiFi network
const char* password = "blablabla";                                        // Password  WiFi network

IPAddress local_IP(192, 168, 1, 80); 
IPAddress gateway(192, 168, 1, 254);
IPAddress subnet(255, 255, 0, 0);
IPAddress primaryDNS(8, 8, 8, 8); //optional
IPAddress secondaryDNS(8, 8, 4, 4); //optional

const char* token = "blablabla";  //Telegram token
// Check the userid with the help of bot @JsonDumpBot or @getidsbot (work also with groups)
// https://t.me/JsonDumpBot  or  https://t.me/getidsbot
int64_t userid = 789512150; //replace with your userid

int wifi_start_counter = 0;
// Timezone definition to get properly time from NTP server
#define MYTZ "WET0WEST,M3.5.0/1,M10.5.0" // Portugal
AsyncTelegram2 myBot(secure_client);

#define LED_PIN 33 // Pin 33 on ESP32CAM

// AI Thinker
#define PWDN_GPIO_NUM 32
#define RESET_GPIO_NUM -1
#define XCLK_GPIO_NUM 0
#define SIOD_GPIO_NUM 26
#define SIOC_GPIO_NUM 27

#define Y9_GPIO_NUM 35
#define Y8_GPIO_NUM 34
#define Y7_GPIO_NUM 39
#define Y6_GPIO_NUM 36
#define Y5_GPIO_NUM 21
#define Y4_GPIO_NUM 19
#define Y3_GPIO_NUM 18
[... 196 lines cut ...]
}

void setup() {
  WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);       // disable brownout detector

  Serial.begin(115200);
  Serial.println();

  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  blinkLED(2, 100); // blink red led  two times on power on

  init_wifi();

  sendToTelegram();

  blinkLED(3, 100); // blink red led  three times when picture has been sent

}

void loop() {

}
```

cotestatnt, 2024-01-29:
Hi @FBMinis 

You are sending photo using this method:
` inline bool sendPhoto(int64_t chat_id, uint8_t *data, size_t size, const char *caption = nullptr)`

As you can see, caption is already supported but actually null as default.
So you can prepare your timestamp string just before sending the message and include it.
```cpp
   char caption_str[20];
   time_t now = time(nullptr);
   tInfo = *localtime(&now);
   strftime(caption_str, sizeof(caption_str), "%d/%m/%Y %H:%M:%S", &tInfo);
   myBot.sendPhoto(userid, fb->buf, fb->len, caption_str);
```

FBMinis, 2024-01-29:
Working perfectly, thank you!

```
  char caption_str[20];
  struct tm tInfo;
  
  time_t now = time(nullptr);
  tInfo = *localtime(&now);
  strftime(caption_str, sizeof(caption_str), "%d/%m/%Y %H:%M:%S", &tInfo);
  myBot.sendPhoto(userid, fb->buf, fb->len, caption_str);
```

FBMinis, 2024-02-23:
Hi @cotestatnt ,

Do we need to wait for a message or photo to finish being sent before putting the ESP32 to deep-sleep?

```
static unsigned long previousMillis = 0;
unsigned long interval = 10000UL; // 30 seconds in milliseconds

void camera_send() {

   unsigned long currentMillis = millis();
  // Send a picture after 30 seconds
  if (currentMillis - previousMillis >= interval) {
    Serial.println("30 seconds have passed");
    previousMillis = currentMillis;

    Serial.println("\nSending Photo from CAM");
    if (sendPicture(userid)) {
      Serial.print("Picture sent successfully ");
      Serial.println(millis());
      GoToSleep();
      // hibernate(); // set rtc alarm and turn off
    } else {
      myBot.sendTo(userid, "Error, picture not sent.");
      GoToSleep();
      // hibernate(); // set rtc alarm and turn off
    }
  }

}

void GoToSleep() {

  //  int sleep_countdown = 0; // not sure if needed
  //  while (1) {
  //    delay(500);
  //    Serial.print(".");
  //    sleep_countdown++;
  //    if (sleep_countdown >= 10) { // Adjusted for 5 seconds with a delay of 500 milliseconds
  //      Serial.println("");
  //      break; // Exit the loop after executing GoToSleep()
  //    }
  //  }

  Serial.print("Entering deep sleep mode at ");
  Serial.println(millis());
  Serial.flush();
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  esp_deep_sleep_start();

}
```