cotestatnt/AsyncTelegram2 #138: Certificate validation and expiry

moldabekov, 2024-04-17:
Hello!

First off, excellent work. My ESP8266 is running much quicker with your TG bot library. However, I'm quite concerned about the embedded TLS certificate for api.telegram.org. I understand it's likely in place to reduce the overhead for connection time. But what occurs when the certificate expires, which is expected in September 2024?

I am planning to put my device in remote distant place, and now thinking what workarounds or fixes can be applied to avoid any service interruptions.

zenbooster, 2024-05-13:
> Hello!
in September 2024?

Ha-ha!! ))
```
Unable to connect to Telegram server
(SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
(SSLClient)(SSL_ERROR)(m_print_br_error): Certificate is expired or not yet valid.

Unable to connect to Telegram server
(SSLClient)(SSL_ERROR)(m_start_ssl): Failed to initlalize the SSL layer
(SSLClient)(SSL_ERROR)(m_print_br_error): Certificate is expired or not yet valid.
```

moldabekov, 2025-08-03:
@fmarzocca, @zenbooster

I have updated embedded certificate in [my pull request](https://github.com/cotestatnt/AsyncTelegram2/pull/158)

moldabekov, 2025-08-04:
@fmarzocca, while @cotestatnt is silent, what you can do is to get newer certificate from updated [tg_certificate.h](https://github.com/moldabekov/AsyncTelegram2/blob/master/src/tg_certificate.h) from my branch and recompile your application again.