cotestatnt/AsyncTelegram2 #142: Bot disconnects after few minutes

noname453, 2024-06-14:
I am using https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/lightBot/lightBot.ino to test my bot and everything works fine for few minutes, the led responds to my on/off commands,  but than my serial monitor prints "Unable to connect to Telegram server" over and over again.
The device that I'm using is AI THINKER esp32-cam.
Looked at #13 which sounds similar, but I am using the example as-is, so my loop does have the line `myBot.getNewMessage(msg)`.
Also tried to set https://github.com/cotestatnt/AsyncTelegram2/blob/main/src/AsyncTelegram2.h#L20 (FS_SUPPORT) to `TRUE`, but there is no extra output in my console, all I get is:

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1448
> load:0x40078000,len:14844
> ho 0 tail 12 room 4
> load:0x40080400,len:4
> load:0x40080404,len:3356
> entry 0x4008059c
> 
> Starting TelegramBot...
> 
> Test Telegram connection... OK
> Bot name: @esp32bot
> 
> 
> Unable to connect to Telegram server

I've managed to find that the last line is the output of AsyncTelegram2::checkConnection, but this is as far as I could get.
Any suggestions?

fmanthey, 2024-06-15:
Facing a similar issue using [echoBot_wifiNINA](https://github.com/cotestatnt/AsyncTelegram2/blob/master/examples/echoBot_wifiNINA/echoBot_wifiNINA.ino) example code as is. Just inserted my Wifi credentials, telegram token & userid. I am using [Arduino Nano 33 IoT](https://store.arduino.cc/products/arduino-nano-33-iot) as a platform.

Bot works fine for few minutes and echoes back all messages as expected, but then stops doing so. To figure out where it is hanging I set `#define DEBUG_ENABLE 1` in [AsyncTelegram2.h](https://github.com/cotestatnt/AsyncTelegram2/blob/master/src/AsyncTelegram2.h). It keeps trying to reconnect via AsyncTelegram2::checkConnection() but fails to do so.

`20:10:17.170 -> Unable to connect to Telegram server
20:10:17.170 -> 348919 [AsyncTelegram2.cpp:64 reset()]	Restart Telegram connection
20:10:17.170 -> 
20:10:17.170 -> 348920 [AsyncTelegram2.cpp:22 checkConnection()]	Start handshaking...
20:10:27.193 -> 
20:10:27.193 -> 
20:10:27.193 -> Unable to connect to Telegram server
20:10:27.193 -> 358929 [AsyncTelegram2.cpp:64 reset()]	Restart Telegram connection
20:10:27.193 -> 
20:10:27.193 -> 358929 [AsyncTelegram2.cpp:22 checkConnection()]	Start handshaking...
20:10:37.197 -> 
20:10:37.197 -> 
20:10:37.197 -> Unable to connect to Telegram server
20:10:37.197 -> 368939 [AsyncTelegram2.cpp:64 reset()]	Restart Telegram connection
20:10:37.197 -> 
20:10:37.197 -> 368940 [AsyncTelegram2.cpp:22 checkConnection()]	Start handshaking...`

Any idea what is going on/wrong here?

Thanks & best regards

noname453, 2024-06-19:
Enabled debugging as @fmanthey did, I also get this line in my log:

>Mon Jun 17 17:02:08 2024 [AsyncTelegram2.cpp:199 getUpdates()]  Connection closed from server
>Mon Jun 17 17:02:09 2024 [AsyncTelegram2.cpp:22 checkConnection()]      Start handshaking...

so it looks like the server closes the connection.
Does it have anything to do with the bot's settings?