cotestatnt/AsyncTelegram2 #144: Compilation warning upon Json lib's update

woodlist, 2024-09-27:
`c:\Users\woodl\Documents\Arduino\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp: In member function 'MessageType AsyncTelegram2::getNewMessage(TBMessage&)':
c:\Users\woodl\Documents\Arduino\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp:267:35: warning: 'bool ArduinoJson::V720PB22::JsonDocument::containsKey(TChar*) const [with TChar = const char]' is deprecated: use doc["key"].is<T>() instead [-Wdeprecated-declarations]
  267 |         if (!updateDoc.containsKey("result"))
      |              ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:33,
                 from c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from c:\Users\woodl\Documents\Arduino\libraries\AsyncTelegram2\src\AsyncTelegram2.h:7,
                 from c:\Users\woodl\Documents\Arduino\libraries\AsyncTelegram2\src\AsyncTelegram2.cpp:1:
c:\Users\woodl\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:157:8: note: declared here
  157 |   bool containsKey(TChar* key) const {
      |        ^~~~~~~~~~~`

cotestatnt, 2024-09-27:
Yes I am aware of this.
I have already [committed ](https://github.com/cotestatnt/AsyncTelegram2/commit/e7ba972c69d9d203927b5be0330d794595cf878f#diff-6ab33ec1a74627c0fb9c2c577a60283c8457c341811a8af43cdfae45875a2f85R267) to solve the warning.

woodlist, 2024-09-27:
Thanks! The warning gone.

woodlist, 2024-10-15:
Greetings, Mr. Tolentino.

I would like to report that "keyboard.ino" does not work anymore (The compilation is error free). I suppose that happened upon Json update.