cotestatnt/AsyncTelegram2 #149: Missing library dependency, but it's installed

bartgrefte, 2025-03-22:
[image]

[image]

If I hit install all, this is the log:

> Already installed ArduinoJson@7.3.1
> Downloading AsyncTelegram2@2.3.2
> AsyncTelegram2@2.3.2
> Installing AsyncTelegram2@2.3.2
> Installed AsyncTelegram2@2.3.2

Is this library looking for a specific version of ArduinoJson or is this caused by something else?
Arduino IDE 2.3.4

cotestatnt, 2025-03-22:
I know the message is annoying, but it doesn't depend on the library. 
When there are dependencies specified in a library properties file, the popup always shows up even if the dependencies are already satisfied. You can ignore the warning and proceed with `Install Without Dependencies`

bartgrefte, 2025-03-22:
The notification specifically says it's currently not installed, that makes me think there's an actual check. I hit the install all anyway and the log indicated it was already installed. So the notification is pointless?

cotestatnt, 2025-03-22:
Yes it is.
You can experience the same behavior on many other libraries where dependencies are explicitly indicated.

For example, in the most common Adafruit libraries there are always dependencies of "internal" libraries such as "Adafruit BusIO" or "Adafruit Unified Sensor" and every time the message appears, even if you are just updating.

closed 2025-04-27 by cotestatnt