cotestatnt/AsyncTelegram2 #156: Same library is not added to PlatformIO as is requested from GitHub?

dannyz2, 2025-05-22:
I have a few ESP32 projects that use ASyncTelegram2. 

in my platformio.ini file I have 

```c++
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

build_flags = -D ELEGANTOTA_USE_ASYNC_WEBSERVER=1

lib_deps = 
   	ayushsharma82/ElegantOTA@3.1.6
;	witnessmenow/UniversalTelegramBot@1.3.0
	paulstoffregen/OneWire@2.3.8
	milesburton/DallasTemperature@3.11.0
;    bblanchon/ArduinoJson@6.21.5                    ; Older Version to keep UniversalTelegramBot from Complaining
;	bblanchon/ArduinoJson@7.2.0
    adafruit/Adafruit GFX Library@1.11.5
    adafruit/Adafruit SSD1306@2.5.7
    cotestatnt/AsyncTelegram2@2.3.2
``` 

Note the last line ASyncTelegram2@2.3.2,  the exact version which is current on GitHub.
However, when Platformio imports the library, at least the code to delete a message(Line 405 in ASyncTelegram2.h,) is missing from the library. 

```c++
bool deleteMessage(int64_t chat_id, int32_t message_id);
``` 

I have added the missing lines to any of the projects I use, so that I could use deleteMessage(), but, why is it not there in the first place? 2.3.2 is what is added to PlatformIO project, but, it is different from what is on Github?

cotestatnt, 2025-05-22:
The method `bool deleteMessage(int64_t chat_id, int32_t message_id);` was added with the latest commit, but is not yet available in any release.

I'm going to release the version 2.3.3 with latest commits so if you update you will also find the method in question.

dannyz-2, 2025-05-23:
Ok, thanks, I see you also fixed the missing white space on line 145 in the.cpp file. 
Thanks again for the quick response.